cyclotron-bonn/beamposition_controller #2: DAC-IC

bedakno, 2023-01-03:
Due to the Arduino Nano not having a DAC on its own and the Arduino Due, which was used in the prototype, the idea is to use an IC carrying a DAC with multiple channels.
As there are a lot of possible choices, this is the place to save ideas, ICs which could be used and requirements the IC must fulfil.
The IC should be available at [Mouser](www.mouser.de) so the website's link to the IC should be attached.

bedakno, 2023-01-25:
https://www.mouser.de/new/texas-instruments/ti-dac70504-dac80504-dacs/
DAC-IC that could be used. Features 50MHz SPI Communication, 5V Output, Gain factor setting, 4 channels, samplingrate of around 1MHz.

leloup314, 2023-01-25:
In principle the IC looks good but tke a look at the package: it is exactly what we talked about. Very difficult to solder manually so I would recommend taking a different footprint

leloup314, 2023-01-29:
This is a nice DAC https://www.mouser.de/ProductDetail/Analog-Devices/AD5686RBRUZ?qs=NmRFExCfTkENVENei53ArQ%3D%3D

leloup314, 2023-01-29:
We can also think about having an external 4 Ch DAC **and** an external 8 Ch ADC, both with 16 bit resolution and SPI interface.
Then the Nano simply just does the communication via SPI (which is quite fast) and the calculation. Would probably be a nicer solution.
One could use such an ADC https://www.mouser.de/ProductDetail/Maxim-Integrated/MAX1168BCEG%2b?qs=d3kjJb5aE2YvByMCteG6qw%3D%3D

bedakno, 2023-04-06:
After browsing again myself, I find the same/similar chips as suggested.
I would propose using following ICs.
DAC: [AD5686BRUZ](https://www.mouser.de/ProductDetail/Maxim-Integrated/MAX1168BCEG%2b?qs=d3kjJb5aE2YvByMCteG6qw%3D%3D)
- 4 channel, 16-bit, SPI, 25MS/s

ADC: [MAX1168BCEG+](https://www.mouser.de/ProductDetail/Analog-Devices/AD5686BRUZ?qs=sGAEpiMZZMtgJDuTUz7Xu3joryijS1UxPnReS72fpQQ%3D)
- 8 channel, 16-bit, SPI, 200kS/s

leloup314, 2023-08-14:
- https://www.mouser.de/ProductDetail/Microchip-Technology/MCP3464R-E-ST?qs=W%2FMpXkg%252BdQ5bjobVCBkOgQ%3D%3D
- https://www.mouser.de/ProductDetail/Texas-Instruments/ADS130B04QPWRQ1?qs=Rp5uXu7WBW%2F83WRIW0ul1A%3D%3D

bedakno, 2023-08-18:
I think the [ADS130B04-Q1](https://www.mouser.de/ProductDetail/Texas-Instruments/ADS130B04QPWRQ1?qs=Rp5uXu7WBW%2F83WRIW0ul1A%3D%3D) from Texas Instruments is a good choice.

bedakno, 2023-08-18:
Currently, one can choose between 10-10000nA as full-scale for the IU-conversion. As this won't change, I propose a 120kΩ resistor as 'gain'. 10µA would this way use the full range of the ADC (1.2V) and 10nA would use ~2/3 of the gain 2 setting (±18.75mV). Using this setup would also allow negative beam currents if necessary. Otherwise we could half the gain and shift the incoming voltage to use the full ADC voltage-range. I don't think this is needed as the resolution of 16-bit is probably still far greater than the effective resolution.

leloup314, 2023-08-18:
Sounds good to me :) ICs should be ordered next week

bedakno, 2023-09-13:
In the data sheet of the ADS130B04 I noticed, that AGND to DGND should be between -0.3 and 0.3V. Is that an issue? I remember we discussed that the analog ground should be -1.65V and AVDD 1.65V so that a bipolar signal can be digitised?

leloup314, 2023-09-13:
Well-spotted! This will be a problem. Luckiy, the ADS130B04 states that either analog input can be the following
`Analog input voltage | AINxP, AINxN | min: AGND – 1.6 , max: AVDD + 0.3 V|`